real-stanford/universal_manipulation_interface #2: Link to SLAM marker access restricted

sirwart, 2024-02-21:
I'm trying to find the data collection instructions (https://swanky-sphere-ad1.notion.site/UMI-Data-Collection-Tutorial-4db1a1f0f2aa4a2e84d9742720428b4c?pvs=4) and it has a link to a aruco marker to help calibrate (https://file.notion.so/f/f/9bd858e8-8338-4b7d-9a7e-e005b4302446/53114102-4cdd-407c-a10a-d230b1adfbbd/table_marker.pdf?id=245a0cf6-9d3a-4c3d-874e-a53e124bc6be&table=block&spaceId=9bd858e8-8338-4b7d-9a7e-e005b4302446&expirationTimestamp=1708552800000&signature=2jjsgMCRq3SjR0vJyZIMz9dfBqo947ISo0-NnSDUQQc&downloadName=table_marker.pdf). When I click this link I get the following error:

[image]

zhenjia-xu, 2024-02-21:
Here is the new link: https://umi-gripper.github.io/share/table_marker.pdf
Sorry for the inconvenience.

sirwart, 2024-02-21:
No problem, thanks for the quick fix!

closed 2024-02-21 by sirwart