real-stanford/universal_manipulation_interface #3: what is mean of dist_to_center ?

marcuswuh, 2024-02-22:
I get some erros .

Terminal output:
``` bash
Traceback (most recent call last):
  File "/home/marcuswu/prj/universal_manipulation_interface/scripts/calibrate_slam_tag.py", line 117, in <module>
    main()
  File "/home/marcuswu/miniforge3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/marcuswu/miniforge3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/home/marcuswu/miniforge3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/marcuswu/miniforge3/envs/umi/lib/python3.9/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/marcuswu/prj/universal_manipulation_interface/scripts/calibrate_slam_tag.py", line 94, in main
    all_slam_tag_pos = all_tx_slam_tag[:,:3,3]
IndexError: too many indices for array: array is 1-dimensional, but 3 were indexed
```

in file slam_stdout.txt:
```slam_stdout.txt
Mask img size mismatch! Converting [2704 x 2028] to [960 x 720]

ORB-SLAM3 Copyright (C) 2017-2020 Carlos Campos, Richard Elvira, Juan J. Gómez, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
ORB-SLAM2 Copyright (C) 2014-2016 Raúl Mur-Artal, José M.M. Montiel and Juan D. Tardós, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular-Inertial
Loading settings from /ORB_SLAM3/Examples/Monocular-Inertial/gopro10_maxlens_fisheye_setting_v1_720.yaml
	-Loaded camera 1
	-Loaded image info
	-Loaded IMU calibration
	-Loaded ORB settings
	-Loaded viewer settings
	-Loaded Atlas settings
	-Loaded misc parameters
----------------------------------
SLAM settings: 
	-Camera 1 parameters (Kannala-Brandt): [ 282.907 282.907 480 360 -0.0174584 -0.0280825 0.0128327 -0.0045198 ]
	-Original image size: [ 960 , 720 ]
	-Current image size: [ 960 , 720 ]
	-Sequence FPS: 60
	-Gyro noise: 0.0015
	-Accelerometer noise: 0.017
	-Gyro walk: 5e-05
	-Accelerometer walk: 0.0055
	-IMU frequency: 200
	-Features per image: 1250
	-ORB scale factor: 1.2
	-ORB number of scales: 8
	-Initial FAST threshold: 20
	-Min FAST threshold: 7

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Initialization of Atlas from file: /map/map_atlas.osa
Starting to read the save binary file
End to load the save binary file
Change to map with id: 0
Atlas loaded!
There are 1 cameras in the atlas
Camera 0 is fisheye
Discard points further than 20 m from current camera
Video opened using backend FFMPEG
There are 533 frames in total
video FPS 59.9401
line 1873 keyframes 3MapPoints302
line 1877
mlFrameTimes.size()= 0
KeyFrame::nNextId=6
Relocalized!!
INIT_RELOCALIZE success!
vpKFs.back()->mpImuPreintegrated0x562f1fe18410
pKFCur->mTimeStamp0
vpKFs.front()->mTimeStamp-0.283608
vpKFs.back()->mTimeStamp-0.016675
pKFcur->GetMapPointMatches().size()5006
1691 pKFcur->GetVelocity() 0
0
0
pKFcur->mnId6
Video FPS: 59.9401
ORB-SLAM 3 running at: 21.7047 FPS
cap.read failed!
Shutdown
mpLocalMapper is not finished
mpLoopCloser is not finished

Saving camera trajectory to /data/camera_trajectory.csv ...
There are 1 maps in the atlas
  Map 0 has 7 KFs

CSV camera trajectory saved!
```
I really need some help!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

JinmingM, 2024-02-22:
Hello, how was your dataset obtained? you collect it yourself?

cheng-chi, 2024-02-26:
@marcuswuh It seems like something is wrong with your video file. The line "cap.read failed!" means that OpenCV failed to read the video file.