real-stanford/universal_manipulation_interface #8: Gripper for Franka

Yingdong-Hu, 2024-02-27:
Hi @cheng-chi, this work is incredible! 
I would like to ask if the UMI is designed to be used with the Schunk WSG-50 gripper? 
I want to reproduce UMI on a Franka robot, does that mean I have to buy a WSG-50 gripper and replace Franka's original gripper with the WSG-50?

ErinZhang1998, 2024-03-02:
Hi @cheng-chi, 

For the CAD model you provided, does this mean we won't have the exact same view as the one collected by UMI (because there is no side mirros)?
The CAD model is only for the gopro and finger mounts?

cheng-chi, 2024-03-03:
@ErinZhang1998 The CAD model provided in this thread is a prototype used in earlier stages of this project and is not fully compatiable with UMI. You can modify it to be compatible with UMI, or modify the [Franka mount CAD](https://cad.onshape.com/documents/10e70b8dbf6f9fe17c716739) to be compatible with Franka hand.

closed 2024-03-19 by cheng-chi

HM102, 2024-04-09:
any update on the gripper controller @Yingdong-Hu ?