real-stanford/universal_manipulation_interface #11: GoPro Hero10 replace GoPro Hero 9

cjlong1213, 2024-03-04:
Hi @cheng-chi ,

Thanks for sharing the amazing work!
Can we use **GoPro Hero 10** replace **GoPro Hero 9** since we can not order GoPro Hero?

Thanks

cheng-chi, 2024-03-04:
@cjlong1213 Yes, GoPro Hero 10 should work out of the box!

lakomchik, 2024-03-04:
Hi @cheng-chi.

Is it also possible to use **GoPro Hero 11** or **GoPro Hero 12** instead of  **GoPro Hero 9**?

cheng-chi, 2024-03-04:
@lakomchik I haven't tried on Hero 11 and hero 12 yet. While the code should work for them, you will likley need calibrating camera and IMU parameteres using [OpenImuCameraCalibrator](https://github.com/urbste/OpenImuCameraCalibrator). Hero 11 and 12 have updated sensors and lenses compared to 9 and 10.

lakomchik, 2024-03-05:
@cheng-chi Thank you for detailed answer!

closed 2024-03-19 by cheng-chi