real-stanford/universal_manipulation_interface #18: Error when generating the zarr.zip

elevenjiang1, 2024-03-16:
Hello, thanks a lot for such great work !
I have run the command 
```
python run_slam_pipeline.py example_demo_session
```
However it error when generating the dataset, the errors are shown below:
```
with av.open(vid_args[0][0]) as container:
IndexError: list index out of range
```
[image]

Any solution?

By the way, I find that the Python code style (both in UMI and Diffusion Policy) you write is so fancy and is hard for me to read (though I also always program by Python), I am curious about what class or book you have learned to write such brilliant code style?

jash101, 2024-04-01:
Most likely the path to the example_demo_session folder is wrong. Did you download the example_demo_session in the same folder?

robograffitti, 2024-06-13:
`(umi)$ python run_slam_pipeline.py example_demo_session`

This command download the files in ~/example_demo_session.

Therefore, the following command should be like:
`(umi)$ python scripts_slam_pipeline/07_generate_replay_buffer.py -o ~/example_demo_session/dataset.zarr.zip ~/example_demo_session`

closed 2024-06-15 by elevenjiang1