real-stanford/universal_manipulation_interface #21: How can I deploy umi on other robots？ like jaka robot.

HaMengder, 2024-03-20:


rhobotics, 2024-03-21:
Or at least be able to drive a robot in a simulation like gazebo or omniverse.

back when ALOHA was presented I tried to drive a UR5e through their simulation and ALOHA stack but without any success.

HaMengder, 2024-04-30:
> any updates? bro

i have no wsg gripper,so I removed the code about the wsg gripper. now i can control the robot arm through umi

duidui6666, 2024-05-17:
May I ask if you use jaka's python SDK to implement umi control of the robotic arm? Can I see what you changed? I also use jaka robotic arm, if possible, can I see your code? This is my email 1365351984@qq.com, thank you very much.@HaMengder

fetter1111, 2025-03-30:
> > any updates? bro
> 
> i have no wsg gripper,so I removed the code about the wsg gripper. now i can control the robot arm through umi

I also want to remove the code about  the wsg gripper,but i do not know the gripper's input  dim in the obs_data.If possible, can I see your code? This is my email :1438559781@qq.com .Thank you very much @HaMengder