real-stanford/universal_manipulation_interface #26: Streaming at 2.7K during inference

ErinZhang1998, 2024-03-28:
Dear authors,

I want to stream using the capture card+media mod at 2.7K (same as the training data), because I need to rectify the images using the provided config file. 
But I can only get 1920x1080 images using the capture card at the moment. Could you help?

Thanks!

ErinZhang1998, 2024-03-28:
More specifically, how did you conduct the rectified experiments and what config file did you use to rectify the images during inference time?

Huang-1-Yi, 2024-05-13:
> 尊敬的作者：
> 
> 我想使用 2.7K 的采集卡+媒体模组进行流式传输（与训练数据相同），因为我需要使用提供的配置文件校正图像。但是我目前只能使用采集卡获得 1920x1080 的图像。你能帮忙吗？
> 
> 谢谢！
Hello, I am using an HD60X capture card and a GoPro Hero Black 9, but when I use it as a visual input, it can detect the capture card but shows no signal. Do you have any suggestions? Could you please share your device?

Huang-1-Yi, 2024-05-13:
> Dear authors,
> 
> I want to stream using the capture card+media mod at 2.7K (same as the training data), because I need to rectify the images using the provided config file. But I can only get 1920x1080 images using the capture card at the moment. Could you help?
> 
> Thanks!

Hello, I am using an HD60X capture card and a GoPro Hero Black 9, but when I use it as a visual input, it can detect the capture card but shows no signal. Do you have any suggestions? Could you please share your device?

wut19, 2024-05-21:
> Hello, I am using an HD60X capture card and a GoPro Hero Black 9, but when I use it as a visual input, it can detect the capture card but shows no signal. Do you have any suggestions? Could you [...]
Have you solved this problem? I meet this issue too.

Jingjinganhao, 2024-05-27:
> Have you solved this problem? I meet this issue too.

> Have you solved this problem? I meet this issue too.

Jingjinganhao, 2024-05-27:
> Hello, I am using an HD60X capture card and a GoPro Hero Black 9, but when I use it as a visual input, it can detect the capture card but shows no signal. Do you have any suggestions? Could you [...]

> Have you solved this problem? I meet this issue too.

Huang-1-Yi, 2024-05-28:
The issue has been resolved; a media mod is needed, specifically the GoPro media mod, as it is indeed lacking output.

Jingjinganhao, 2024-05-29:
> The issue has been resolved; a media mod is needed, specifically the GoPro media mod, as it is indeed lacking output.

Thank you very much. I found out that elgato HD60X does not support ubuntu.Apart from using GoPro media mod, do I need to install anything for elgato?