real-stanford/universal_manipulation_interface #33: polymetis server issue

HM102, 2024-04-04:
running the file `scripts_real/launch_franka_interface_server.py` on the NUC returns the following the error:

```
raise _InactiveRpcError(state)
grpc._channel._InactiveRpcError: <_InactiveRpcError of RPC that terminated with:
        status = StatusCode.CANCELLED
        details = "Robot context not valid when calling GetRobotClientMetadata!"
        debug_error_string = "{"created":"@1712241739.450975495","description":"Error received from peer ipv4:127.0.0.1:50051","file":"src/core/lib/surface/call.cc","file_line":952,"grpc_message":"Robot context not valid when calling GetRobotClientMetadata!","grpc_status":1}"
```

should I run  the `launch_robot.py` on polymetis side first?

fighting-j, 2024-04-28:
Hello, I also encountered this problem when using franka. How did you solve it in the end? @HM102

ErinZhang1998, 2024-05-17:
Run this before starting the server `launch_robot.py robot_client=franka_hardware` in `fairo/polymetis`