real-stanford/universal_manipulation_interface #34: Flange position and rotation settings

Yingdong-Hu, 2024-04-08:
[image]
[image]

I'm wondering how to measure these values

JinmingM, 2024-04-25:
Hello,
Have you figured this out yet? Why doesn’t the UR5 need any position and rotation settings while the Franka does? 

Do other robotic arms require similar configurations?

fighting-j, 2024-11-18:
Hello, I have the same question, is there any follow-up? I want to deploy it on my own robot arm, but I am not sure how to set these values.