real-stanford/universal_manipulation_interface #48: Any method that I can visualize the ORB SLAM results?

ysyyork, 2024-06-23:
This would be very helpful and more straightforward than pure CSV and other text file formats.

AlphaGotReal, 2024-06-26:
You can probably visualise by passing in enable_gui=true in the ORB_SLAM3 package
```cpp
bool enable_gui = false;
  app.add_flag("-g,--enable_gui", enable_gui);
.
.
.
.
cv::Ptr<cv::aruco::Dictionary> aruco_dict = cv::aruco::getPredefinedDictionary(aruco_dict_id);
  ORB_SLAM3::System SLAM(
    vocabulary, setting, 
    ORB_SLAM3::System::IMU_MONOCULAR, 
    enable_gui, load_map, save_map, // make enable gui ---> true
    aruco_dict, init_tag_id, init_tag_size
  );
 ```
passing that here(03_batch_slam.py) as --enable-gui true should launch the built-in gui tool by ORB SLAM3
```py
cmd = [
      'docker',
      'run',
      '--rm', # delete after finish
      '--volume', str(video_dir) + ':' + '/data',
      '--volume', str(map_mount_source.parent) + ':' + str(map_mount_target.parent),
      docker_image,
      '/ORB_SLAM3/Examples/Monocular-Inertial/gopro_slam',
      '--vocabulary', '/ORB_SLAM3/Vocabulary/ORBvoc.txt',
      '--setting', '/ORB_SLAM3/Examples/Monocular-Inertial/gopro10_maxlens_fisheye_setting_v1_720.yaml',
      '--input_video', str(video_path),
      '--input_imu_json', str(json_path),
      '--output_trajectory_csv', str(csv_path),
      '--load_map', str(map_mount_target),
      '--mask_img', str(mask_path),
      '--max_lost_frames', str(max_lost_frames),
      '--enable-gui', 'true' # or make it a user argument in some config file or command itself
]
  ```
  but before that you also need to enable gui in docker by running the ```xhost +``` or ```xhost +docker``` command in the particular terminal

cynthia-you, 2024-09-27:
> You can probably visualise by passing in enable_gui=true in the ORB_SLAM3 package
> 
> ```c++
> bool enable_gui = false;
>   app.add_flag("-g,--enable_gui", enable_gui);
> .
> .
> .
> .
> cv::Ptr<cv::aruco::Dictionary> aruco_dict = cv::aruco::getPredefinedDictionary(aruco_dict_id);
>   ORB_SLAM3::System SLAM(
>     vocabulary, setting, 
>     ORB_SLAM3::System::IMU_MONOCULAR, 
>     enable_gui, load_map, save_map, // make enable gui ---> true
>     aruco_dict, init_tag_id, init_tag_size
>   );
> ```
> 
> passing that here(03_batch_slam.py) as --enable-gui true should launch the built-in gui tool by ORB SLAM3
> 
> ```python
> cmd = [
>       'docker',
>       'run',
>       '--rm', # delete after finish
>       '--volume', str(video_dir) + ':' + '/data',
>       '--volume', str(map_mount_source.parent) + ':' + str(map_mount_target.parent),
>       docker_image,
>       '/ORB_SLAM3/Examples/Monocular-Inertial/gopro_slam',
>       '--vocabulary', '/ORB_SLAM3/Vocabulary/ORBvoc.txt',
>       '--setting', '/ORB_SLAM3/Examples/Monocular-Inertial/gopro10_maxlens_fisheye_setting_v1_720.yaml',
>       '--input_video', str(video_path),
>       '--input_imu_json', str(json_path),
>       '--output_trajectory_csv', str(csv_path),
>       '--load_map', str(map_mount_target),
>       '--mask_img', str(mask_path),
>       '--max_lost_frames', str(max_lost_frames),
>       '--enable-gui', 'true' # or make it a user argument in some config file or command itself
> ]
> ```
> 
> but before that you also need to enable gui in docker by running the `xhost +` or `xhost +docker` command in the particular terminal window

Hello, could you specific the path of :
"bool enable_gui = false;
  app.add_flag("-g,--enable_gui", enable_gui);
.
.
.
.
cv::Ptr<cv::aruco::Dictionary> aruco_dict = cv::aruco::getPredefinedDictionary(aruco_dict_id);
  ORB_SLAM3::System SLAM(
    vocabulary, setting, 
    ORB_SLAM3::System::IMU_MONOCULAR, 
    enable_gui, load_map, save_map, // make enable gui ---> true
    aruco_dict, init_tag_id, init_tag_size
  );"

thanks~

AlphaGotReal, 2024-09-29:
The parameter variable actually exists inside the docker container running ORBSLAM3, visit https://github.com/cheng-chi/ORB_SLAM3/blob/master/Examples/Monocular-Inertial/gopro_slam.cc, where you'll find the argument, for the changes to take action you'll have to build the package again inside the docker container.
However you can control the param within the python script in scripts_slam_pipeline/03_batch_slam.py where you append the argument 
```py
'--enable-gui', 'true' # or make it a user argument in some config file or command itself
```
at line number 124 in the `cmd` list. 
This should run the GUI, but before this you'll have to run the `xhost +docker` command in the particular terminal window so as to give permission for the docker container to run apps that include GUI.

fighting-j, 2024-10-29:
I followed the above method and added '--enable-gui', 'true' to scripts_slam_pipeline/03_batch_slam.py, but the following problem occurred：
[image]
How can I solve it

nazir-hk, 2024-11-04:
> I followed the above method and added '--enable-gui', 'true' to scripts_slam_pipeline/03_batch_slam.py, but the following problem occurred： ![截图 2024-10-29 10-53-21](https://private-user-images.githubusercontent.com/87596844/380954349-f62d1a4c-a434-4f1a-b258-bf35200a94a6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzA3MTUwMDIsIm5iZiI6MTczMDcxNDcwMiwicGF0aCI6Ii84NzU5Njg0NC8zODA5NTQzNDktZjYyZDFhNGMtYTQzNC00ZjFhLWIyNTgtYmYzNTIwMGE5NGE2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDQlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA0VDEwMDUwMlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTc2ZjY2MWY4M2YyOWJkMzBkMThjNzcwN2JiY2VhYTllZTdmYjVkMmM5YmFlZGQ1NmMyZjMwZDM1NWQwMmJiYTQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.1apLEX9O7F0RA2qIsbUNIdCC9pckVws5oQ124db1u_E) How can I solve it

Hi, were you able to resolve this?

AlphaGotReal, 2024-11-16:
> I followed the above method and added '--enable-gui', 'true' to scripts_slam_pipeline/03_batch_slam.py, but the following problem occurred： ![截图 2024-10-29 [...]

does the gui for orb slam not show up? The print statement is just a log and nothing much ig