real-stanford/universal_manipulation_interface #51: Is there something I miss? Failed to run slam.

Kimagure7, 2024-07-02:
I was trying to run `python run_slam_pipeline.py example_demo_session` after i download example data. And an error occurred:
`(umi) zhaozy@rykj-002:~/ZoeyChiu/universal_manipulation_interface$ python run_slam_pipeline.py example_demo_session
############## 00_process_videos #############
Traceback (most recent call last):
  File "/home/zhaozy/ZoeyChiu/universal_manipulation_interface/scripts_slam_pipeline/00_process_videos.py", line 122, in <module>
    main()
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/zhaozy/ZoeyChiu/universal_manipulation_interface/scripts_slam_pipeline/00_process_videos.py", line 31, in main
    input_dir.mkdir()
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/pathlib.py", line 1323, in mkdir
    self._accessor.mkdir(self, mode)
FileNotFoundError: [Errno 2] No such file or directory: '/home/zhaozy/ZoeyChiu/example_demo_session/raw_videos'
Traceback (most recent call last):
  File "/home/zhaozy/ZoeyChiu/universal_manipulation_interface/run_slam_pipeline.py", line 120, in <module>
    main()
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/zhaozy/ZoeyChiu/universal_manipulation_interface/run_slam_pipeline.py", line 41, in main
    assert result.returncode == 0
AssertionError`

What should I do?

Kimagure7, 2024-07-02:
(umi) zhaozy@rykj-002:~/ZoeyChiu/universal_manipulation_interface/example_demo_session$ ls
GX010213.MP4  GX010214.MP4  GX010215.MP4  GX010232.MP4  GX010234.MP4  GX010236.MP4  GX010238.MP4

Kimagure7, 2024-07-03:
After I take the example_demo_sessions folder to the parent folder ( at the same level directory as the umi folder).The error is:
(umi) zhaozy@rykj-002:~/ZoeyChiu/universal_manipulation_interface$ python run_slam_pipeline.py example_demo_session
############## 00_process_videos #############
Traceback (most recent call last):
  File "/home/zhaozy/ZoeyChiu/universal_manipulation_interface/scripts_slam_pipeline/00_process_videos.py", line 122, in <module>
    main()
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/zhaozy/ZoeyChiu/universal_manipulation_interface/scripts_slam_pipeline/00_process_videos.py", line 47, in main
    shutil.move(max_path, mapping_vid_path)
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/shutil.py", line 825, in move
    os.rename(src, real_dst)
TypeError: rename: src should be string, bytes or os.PathLike, not NoneType
Traceback (most recent call last):
  File "/home/zhaozy/ZoeyChiu/universal_manipulation_interface/run_slam_pipeline.py", line 120, in <module>
    main()
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/data/zhaozy/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/zhaozy/ZoeyChiu/universal_manipulation_interface/run_slam_pipeline.py", line 41, in main
    assert result.returncode == 0
AssertionError

Kimagure7, 2024-07-03:
I have solved it.
You need to put all the mp4 into example_demo_session/raw_videos and make sure example_demo_session folder is  at the same level directory as the umi folder.
This is not mentioned in the readme.md.You should update it.

btw, docker is needed. In 02, code will pull a docker image

Kimagure7, 2024-07-03:
however,in ` python scripts_slam_pipeline/07_generate_replay_buffer.py -o example_demo_session/dataset.zarr.zip example_demo_session`
example_demo_session is in the umi folder
.........

adamphung-ho, 2024-08-14:
> I have solved it. You need to put all the mp4 into example_demo_session/raw_videos and make sure example_demo_session folder is at the same level directory as the umi folder. This is not mentioned in the readme.md.You should update it.
> 
> btw, docker is needed. In 02, code will pull a docker image

I put all videos in the folder example_demon_session/raw_videos as you suggested, but it didn't work. Can you explain more how to solve it ?

Kimagure7, 2024-08-14:
> I put all videos in the folder example_demon_session/raw_videos as you suggested, but it didn't work. Can you explain more how to solve it ?

That's all I have done😂.
Command ls should look like this:
- example...
 -- raw_videos...
   ---.mp4
-universal....

maybe you can use debug mode to run the code and,use breakpoint to check whether the path is right.

adamphung-ho, 2024-08-14:
> That's all I have done😂. Command ls should look like this:
>
> * example...
>   -- raw_videos...
>   ---.mp4
>   -universal....
>
> maybe you can use debug mode to run the code and,use breakpoint to check [...]

I’ve resolved the issue. I had mistakenly set the wrong path for the example_demo_session folder. Thank you for your suggestion.