real-stanford/universal_manipulation_interface #52: Is there a guide for collecting bimanual data?

ysyyork, 2024-07-07:
I followed the data collection guide [here](https://swanky-sphere-ad1.notion.site/UMI-Data-Collection-Instruction-4db1a1f0f2aa4a2e84d9742720428b4c) and I was able to successfully replicate single hand data collection. There is a step for mapping and gripper calibration. I'm not too sure how the code handles two arm cases for this cus I thought for single arm, the code just check the oldest file as the mapping and gripper calibration video file and process it. When with bimanual setup, does the data collection order matters? is there any concrete guide for that? also, for training, is there a specific config file I should use? Thanks.

mrhouse-sweet, 2024-07-08:
When you run this example “python eval_real.py --robot_config=example/eval_robots_config.yaml -i cup_wild_vit_l.ckpt -o data/eval_cup_wild_example
”, have you encountered the following errors? How can you solve them? I have tried online methods, but they have all failed

ysyyork, 2024-07-08:
> When you run this example “python eval_real.py --robot_config=example/eval_robots_config.yaml -i cup_wild_vit_l.ckpt -o data/eval_cup_wild_example ”, have you encountered the following errors? How can you solve them? I have tried online methods, but they have all failed
> 
> error： Traceback (most recent call last): File "/home/ubuntu/universal_manipulation_interface-main/eval_real.py", line 59, in from umi.real_world.spacemouse_shared_memory import Spacemouse File "/home/ubuntu/universal_manipulation_interface-main/umi/real_world/spacemouse_shared_memory.py", line 4, in from spnav import spnav_open, spnav_poll_event, spnav_close, SpnavMotionEvent, SpnavButtonEvent ModuleNotFoundError: No module named 'spnav'

it clearly said no module name spnav? you should install spnav then but this is not relevant to my question and I'm not the maintainer. you should ask some place else.

mrhouse-sweet, 2024-07-09:
I have installed SPNAV but still reported this error. Thank you, I will ask someone else again

ysyyork, 2024-07-10:
can someone help on this? thanks

closed 2024-07-12 by ysyyork