real-stanford/universal_manipulation_interface #58: Unable to read camera feed from GoPro

HZWang96, 2024-08-29:
Hi @cheng-chi,

I have connected my GoPro camera to the Elgato capture card (micro HDMI from GoPro media mod -> 'HDMI in' on Elgato capture card) and connected the Elgato capture card to my workstation PC (USB type C from Elgato capture card -> USB 3.0 on workstation PC). I have checked that I able to get an output from the Elgato capture card (I attached an HDMI cable to 'HDMI out' and displayed the output to my monitor). However, I am unable to read the output from the Elgato capture card on my workstation PC (OpenCV window is black).

I have also confirmed that my workstation PC is able to detect the Elgato capture card (the path displayed is '/dev/v4l/by-id/usb-Elgato_Elgato_HD60_X_A00XB32721NNWB-video-index0').

Do you know what might be the source of the problem? Thanks in advance for your help!

fighting-j, 2024-10-15:
Hi, I have the same problem, have you solved it now? @HZWang96

WilliamBonilla62, 2024-10-17:
Hey did fix your issue ?

In my case I have the image once every 20 times we try.

WilliamBonilla62, 2024-10-18:
Hi @HZWang96, 

I resolved the issue you have to change the following line in **umi/real_world/uvc_camera.py** : 

from : 

       `examples['camera_capture_timestamp'] = 0.0
        examples['camera_receive_timestamp'] = 0.0
        examples['timestamp'] = 0.0
        examples['step_idx'] = 0

        vis_ring_buffer = SharedMemoryRingBuffer.create_from_examples(
            shm_manager=shm_manager,
            examples=examples if vis_transform is None 
                else vis_transform(dict(examples)),
            get_max_k=1,
            get_time_budget=0.2,
            put_desired_frequency=capture_fps
        )`
        
 to : 
 
      `examples['camera_capture_timestamp'] = 0.0
        examples['camera_receive_timestamp'] = 0.0
        examples['timestamp'] = 0.0
        examples['step_idx'] = 0

        vis_examples = copy.deepcopy(examples)

        if vis_transform is not None:
            vis_shape = (720, 960, 3)        
        else:
            vis_shape = shape+(3,)
        vis_examples['color'] = np.empty(shape=vis_shape, dtype=np.uint8)
        vis_ring_buffer = SharedMemoryRingBuffer.create_from_examples(
            shm_manager=shm_manager,
            examples=vis_examples,
            get_max_k=1,
            get_time_budget=0.2,
            put_desired_frequency=capture_fps
        )`