real-stanford/universal_manipulation_interface #59: I want to know how you convert the EE pose into the 6d action of the arm.

noooob-coder, 2024-09-13:
My current research work is based on your article. I read your paper, and the model in the paper generates the EE relative posture of the gripper. However, the paper does not mention how to convert the relative EE posture into 6d action. I would be grateful if you could provide reference articles or relevant code.

lakomchik, 2024-09-19:
@noooob-coder FIrst, end effector poses converted from the world [global frame](https://github.com/real-stanford/universal_manipulation_interface/blob/main/diffusion_policy/dataset/umi_dataset.py#L306) to [relative](https://github.com/real-stanford/universal_manipulation_interface/blob/main/diffusion_policy/dataset/umi_dataset.py#L316) frame.  Then matrices with relative poses [converted to vector](https://github.com/real-stanford/universal_manipulation_interface/blob/main/diffusion_policy/dataset/umi_dataset.py#L322) where first 3 values are linear transformations in relative frame and other 6 - 6D orientation transforms . Details of convertation from SO3 to 6D you can find [here](https://github.com/real-stanford/universal_manipulation_interface/blob/main/umi/common/pose_util.py#L95).

yangfengzzz, 2025-09-25:
But how to use the relative pose in IK solver? where should I put the robot base for first frame?