real-stanford/universal_manipulation_interface #60: run_slam_pipline alwasy creat map failed

cynthia-you, 2024-09-27:
Hi, I've watched the example_demo_session for thousand times, but my data cannot generate demos*/cam_trajectory.csv after 03_batch_slam.py.
why why why, it drives my crazy, can somebody tell me , huge thanks!!!
[image]
[image]

seann999, 2024-10-02:
You might be going too fast or lacking sufficient viewpoint coverage in your mapping video. It also depends on your task/demos. If you can share your videos via GDrive or smth, might be able to give more pointers.

cynthia-you, 2024-10-08:
> You might be going too fast or lacking sufficient viewpoint coverage in your mapping video. It also depends on your task/demos. If you can share your videos via GDrive or smth, might be able to give more pointers.

#######
Thanks for your kindly help~~ 
https://drive.google.com/file/d/13k_wPeGYCmdU_wp9pKBqQRftK5EisdvF/view?usp=drive_link

paulverdugo, 2025-06-02:
> ####### Thanks for your kindly help~~ https://drive.google.com/file/d/13k_wPeGYCmdU_wp9pKBqQRftK5EisdvF/view?usp=drive_link

Hello cynthia-you, I am having the same issue. Wanted to ask if you found a solution. I would very much appreciate your help. Thanks.