real-stanford/universal_manipulation_interface #62: Custom Robotic Arm

bhargavbompalli, 2024-09-30:
Thanks for your work! I have a 6-axis robotic arm made by uFactory which I want to test this with. 

What are the important changes need to be made in order to make this compatible with the xArm6? Is this even possible?

Thanks!

seann999, 2024-10-02:
You'll need to replace the gripper to a parallel gripper with the same TPU fingers as the UMI. If you can get a WSG-50 gripper, simplest way might be to modify either of the provided mounts to attach to the xArm flange. Rest should be the same. Otherwise do something like UMI on Legs with an ALOHA-type mechanism.

bhargavbompalli, 2024-10-23:
What are some considerations I should take into account if I were to use the xArm parallel gripper instead? Would this require significant code changes?

seann999, 2024-11-18:
You would probably need to modify the UMI hardware to match its fingers, camera, and camera placement