real-stanford/universal_manipulation_interface #64: Is it possible to test in simulation environment?

ruoxianglee, 2024-10-17:
Unfortunately, we can not use the franka panda arm until next few months. 

I am wondering if it is possible to set up a scene and robot with the UMI gripper in a simulation environment. Or are there any other solutions or alternative methods for simulation environent? 

Thanks.

zalan159, 2024-11-17:
I've connected Isaac sim to UMI ,but works not very well. seems like theres some kind of coordinate problem.

George-Gwj, 2025-02-20:
> I've connected Isaac sim to UMI ,but works not very well. seems like theres some kind of coordinate problem.

您好，请问一下您在isaac sim中使用的相机是gopro那种鱼眼相机吗？还是类似于d435那种？

Gellert5225, 2025-04-04:
@zalan159 would you mind sharing the URDF for the UMI? I created my own but the joint motion is kinda bugged.

Gellert5225, 2025-04-24:
I simulated a simple pick and place task with UMI inside NVIDIA Isaac Sim, but yet to implement the diffusion policy.

Will update once I successfully implement the diffusion model. 

https://github.com/user-attachments/assets/6ba7cd31-dd60-4a95-b2fc-72dd1266863c

supershiye, 2025-08-19:
> I simulated a simple pick and place task with UMI inside NVIDIA Isaac Sim, but yet to implement the diffusion policy.
> 
> Will update once I successfully implement the diffusion model.
> recording.mp4

That's cool! How did you set the camera? Just a fish eye?

adrianodelr, 2025-10-29:
> I simulated a simple pick and place task with UMI inside NVIDIA Isaac Sim, but yet to implement the diffusion policy.
> 
> Will update once I successfully implement the diffusion model.
> recording.mp4

Nice setup! Did you have any success with the diffusion policy ?

From the video I assume you modeled the gripper as rigid, right ?

Gellert5225, 2026-01-12:
So I have finally implemented the diffusion policy, and performed a simple pick and drop task. 

Sometimes the gripper misses the block by a tiny bit, which I believe can be improved with more training data. Currently I trained the policy with 200 episodes, 10 fps each. 

The method I used to collect these episodes is VR teleoperation. I did not use the 3d printed UMI + GoPro combo. 

As you can see, the policy does not know when to stop, it just kept on trying to grab over and over again. 

https://github.com/user-attachments/assets/b183d01e-fd6e-4573-adb3-768e854343fc