real-stanford/universal_manipulation_interface #65: Custom camera calibration file

nazir-hk, 2024-10-25:
Hi, I calibrated hero12 using OpenICC now I have an intrinsics json file and a yaml file. json file i can put it under /example/calibration/ and update the code accordingly. The yaml file seems to be required in two scripts: create_map.py and batch_slam.py. And it's path seems within a docker environment? (`/ORB_SLAM3/Examples/Monocular-Inertial/gopro10_maxlens_fisheye_setting_v1_720.yaml`). So how do i provide my custom yaml file?

nazir-hk, 2024-10-25:
ok, i was able to edit the image by cloning it first and adding my calibration file

closed 2024-10-25 by nazir-hk

omarrayyann, 2024-12-04:
@nazir-hk Hello. Are you also using max lens with gopro12? If so, could you share the camera intrinsics

nazir-hk, 2024-12-05:
Hi, yes i'm using max lens mod 2.0 with hero12. 

Yes, I could post my calibration file. But funny thing, I was able to run successful mapping/training on my example dataset with the default gopro10_maxlens file and not with my custom camera calibration file. Weird...

omarrayyann, 2024-12-05:
Interesting. I just calibrated the camera using OpenImuCameraCalibrator

soochulYoo, 2024-12-17:
Hi @nazir-hk , 

How did you solve the create_map problem?

I think the GoPro10 and GoPro12 are different requiring specific calibration yaml file.

I am trying this project with GoPro13 with calibration yaml file using OpenICC.

However, I still failed to create map.

Thank you for your help in advance.

yijiano, 2025-09-22:
> Interesting. I just calibrated the camera using OpenImuCameraCalibrator. Will let you know how it goes

Do you have any updates

> Hi [@nazir-hk](https://github.com/nazir-hk) ,
>
> How did you solve the create_map problem?
>
> I think the GoPro10 and GoPro12 are different requiring specific calibration yaml file.
>
> I am trying this [...]

Also, did you manage to get it working with the GoPro12?

> Hi, yes i'm using max lens mod 2.0 with hero12.
> 
> Yes, I could post my calibration file. But funny thing, I was able to run successful mapping/training on my example dataset with the default gopro10_maxlens file and not with my custom camera calibration file. Weird...

Also, if you don't mind, can I see your calibration file? I am trying to do something similar but I haven't been able to get any good results with the calibration.