real-stanford/universal_manipulation_interface #66: Custom Data SLAM failed

mk2001233, 2024-11-04:
When I execute python run_slam_pipeline.py stack_cubes_v6, I found that I encountered an error when executing 02_create_map.py, which prevented me from getting map_atlas.osa
The log is as follows:
[slam_stderr.txt](https://github.com/user-attachments/files/17615258/slam_stderr.txt)
[slam_stdout.txt](https://github.com/user-attachments/files/17615262/slam_stdout.txt)

nazir-hk, 2024-11-04:
Are you using default yaml, or did you make changes to it? I encountered some similar errors when using my custom yaml file.

WilliamBonilla62, 2024-11-06:
On my side I changed the arruco of the gripper. I had the right gripper arucco and changed for the left one and it worked.

MarioCavero, 2024-11-07:
Can you try running it visually to see what happens with the video? I have encountered that a few times a certain video is being mapped, and after the first keyframe, the video closes (as if corrupt or some error). The output is the same as yours as the .osa map is not created after the mapping process (which exited abruptly). Is this the same case? @mk2001233

LonghaiZHAO, 2024-11-16:
I meet the same error:

############# 02_create_map ###########
Pulling docker image chicheng/orb_slam3:latest
latest: Pulling from chicheng/orb_slam3
Digest: sha256:fbf562a6707a0fe9bafb04375c4cfb7b051156c14bc1e820f3f238b9aa17744f
Status: Image is up to date for chicheng/orb_slam3:latest
docker.io/chicheng/orb_slam3:latest
CompletedProcess(args=['docker', 'run', '--rm', '--volume', '/home/zlh/Desktop/example_demo_session/demos/mapping:/data', '--volume', '/home/zlh/Desktop/example_demo_session/demos/mapping:/map', 'chicheng/orb_slam3:latest', '/ORB_SLAM3/Examples/Monocular-Inertial/gopro_slam', '--vocabulary', '/ORB_SLAM3/Vocabulary/ORBvoc.txt', '--setting', '/ORB_SLAM3/Examples/Monocular-Inertial/gopro10_maxlens_fisheye_setting_v1_720.yaml', '--input_video', '/data/raw_video.mp4', '--input_imu_json', '/data/imu_data.json', '--output_trajectory_csv', '/data/mapping_camera_trajectory.csv', '--save_map', '/map/map_atlas.osa', '--mask_img', '/data/slam_mask.png'], returncode=132)
Traceback (most recent call last):
  File "/home/zlh/Desktop/umi2/run_slam_pipeline.py", line 120, in <module>
    main()
  File "/opt/miniconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/opt/miniconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/opt/miniconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/opt/miniconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/zlh/Desktop/umi2/run_slam_pipeline.py", line 68, in main
    assert map_path.is_file()
AssertionError

seann999, 2024-11-18:
If you share the videos, we might be able to reproduce it on our side?

Ghr0228, 2025-04-14:
> I meet the same error:
> 
> ############# 02_create_map ########### Pulling docker image chicheng/orb_slam3:latest latest: Pulling from chicheng/orb_slam3 Digest: sha256:fbf562a6707a0fe9bafb04375c4cfb7b051156c14bc1e820f3f238b9aa17744f Status: Image is up to date for chicheng/orb_slam3:latest docker.io/chicheng/orb_slam3:latest CompletedProcess(args=['docker', 'run', '--rm', '--volume', '/home/zlh/Desktop/example_demo_session/demos/mapping:/data', '--volume', '/home/zlh/Desktop/example_demo_session/demos/mapping:/map', 'chicheng/orb_slam3:latest', '/ORB_SLAM3/Examples/Monocular-Inertial/gopro_slam', '--vocabulary', '/ORB_SLAM3/Vocabulary/ORBvoc.txt', '--setting', '/ORB_SLAM3/Examples/Monocular-Inertial/gopro10_maxlens_fisheye_setting_v1_720.yaml', '--input_video', '/data/raw_video.mp4', '--input_imu_json', '/data/imu_data.json', '--output_trajectory_csv', '/data/mapping_camera_trajectory.csv', '--save_map', '/map/map_atlas.osa', '--mask_img', '/data/slam_mask.png'], returncode=132) Traceback (most recent call last): File "/home/zlh/Desktop/umi2/run_slam_pipeline.py", line 120, in main() File "/opt/miniconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1157, in **call** return self.main(*args, **kwargs) File "/opt/miniconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1078, in main rv = self.invoke(ctx) File "/opt/miniconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1434, in invoke return ctx.invoke(self.callback, **ctx.params) File "/opt/miniconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 783, in invoke return __callback(*args, **kwargs) File "/home/zlh/Desktop/umi2/run_slam_pipeline.py", line 68, in main assert map_path.is_file() AssertionError

Hi,bro,I meet same question,How to communicate with you? I want to know the source of the problem and how do you deal with it?