real-stanford/universal_manipulation_interface #73: 02_create_map failed!

denjio, 2025-02-11:
Thank you very much for your excellent work. We encountered some small issues while using it. Specifically, we first ran the process with your data and it worked fine, but when we used our own data, we encountered some problems. The issue is reflected in the SLAM mapping process, where the error is as follows:
Traceback (most recent call last):
  File "/home/qianrong/Code/universal_manipulation_interface-main/run_slam_pipeline.py", line 127, in <module>
    main()
  File "/home/qianrong/anaconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/home/qianrong/anaconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx) 
  File "/home/qianrong/anaconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/qianrong/anaconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/home/qianrong/Code/universal_manipulation_interface-main/run_slam_pipeline.py", line 75, in main
    assert map_path.is_file()
AssertionError
We have checked the relevant issues, and the problem is indeed in the mapping process. Could you please help us check if there is any issue with the video? We would greatly appreciate it

denjio, 2025-02-11:
https://drive.google.com/file/d/17_H8Kk4l226zsalHLYCfZSuTcvlHUIhV/view?usp=drive_link

denjio, 2025-02-12:
We have solved this problem, if you have relevant questions welcome to communicate!

LouisGuibert-LS2N, 2025-02-13:
Hello, I just got the same problem, can you tell how do you solve it please.

ssspeg, 2025-02-14:
> 您好，我刚刚遇到了同样的问题，请问您是如何解决的？

We solved the mapping problem by collecting data in environments with better lighting, wider visibility, and more complex background textures.

LouisGuibert-LS2N, 2025-02-14:
I got the same problem with only the demo, not custom videos:

############# 02_create_map ###########
Pulling docker image chicheng/orb_slam3:latest
latest: Pulling from chicheng/orb_slam3
Digest: sha256:fbf562a6707a0fe9bafb04375c4cfb7b051156c14bc1e820f3f238b9aa17744f
Status: Image is up to date for chicheng/orb_slam3:latest
docker.io/chicheng/orb_slam3:latest
CompletedProcess(args=['docker', 'run', '--rm', '--volume', '/home/lguibert/Desktop/UMI_test/universal_manipulation_interface/example_demo_session/demos/mapping:/data', '--volume', '/home/lguibert/Desktop/UMI_test/universal_manipulation_interface/example_demo_session/demos/mapping:/map', 'chicheng/orb_slam3:latest', '/ORB_SLAM3/Examples/Monocular-Inertial/gopro_slam', '--vocabulary', '/ORB_SLAM3/Vocabulary/ORBvoc.txt', '--setting', '/ORB_SLAM3/Examples/Monocular-Inertial/gopro10_maxlens_fisheye_setting_v1_720.yaml', '--input_video', '/data/raw_video.mp4', '--input_imu_json', '/data/imu_data.json', '--output_trajectory_csv', '/data/mapping_camera_trajectory.csv', '--save_map', '/map/map_atlas.osa', '--mask_img', '/data/slam_mask.png'], returncode=132)
Traceback (most recent call last):
  File "/home/lguibert/Desktop/UMI_test/universal_manipulation_interface/run_slam_pipeline.py", line 120, in <module>
    main()
  File "/home/lguibert/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
  File "/home/lguibert/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1082, in main
    rv = self.invoke(ctx)
  File "/home/lguibert/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/lguibert/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 788, in invoke
    return __callback(*args, **kwargs)
  File "/home/lguibert/Desktop/UMI_test/universal_manipulation_interface/run_slam_pipeline.py", line 68, in main
    assert map_path.is_file()
AssertionError

ssspeg, 2025-02-18:
> I got the same problem with only the demo, not custom videos:
> 
> ############# 02_create_map ########### Pulling docker image chicheng/orb_slam3:latest latest: Pulling from chicheng/orb_slam3 Digest: sha256:fbf562a6707a0fe9bafb04375c4cfb7b051156c14bc1e820f3f238b9aa17744f Status: Image is up to date for chicheng/orb_slam3:latest docker.io/chicheng/orb_slam3:latest CompletedProcess(args=['docker', 'run', '--rm', '--volume', '/home/lguibert/Desktop/UMI_test/universal_manipulation_interface/example_demo_session/demos/mapping:/data', '--volume', '/home/lguibert/Desktop/UMI_test/universal_manipulation_interface/example_demo_session/demos/mapping:/map', 'chicheng/orb_slam3:latest', '/ORB_SLAM3/Examples/Monocular-Inertial/gopro_slam', '--vocabulary', '/ORB_SLAM3/Vocabulary/ORBvoc.txt', '--setting', '/ORB_SLAM3/Examples/Monocular-Inertial/gopro10_maxlens_fisheye_setting_v1_720.yaml', '--input_video', '/data/raw_video.mp4', '--input_imu_json', '/data/imu_data.json', '--output_trajectory_csv', '/data/mapping_camera_trajectory.csv', '--save_map', '/map/map_atlas.osa', '--mask_img', '/data/slam_mask.png'], returncode=132) Traceback (most recent call last): File "/home/lguibert/Desktop/UMI_test/universal_manipulation_interface/run_slam_pipeline.py", line 120, in main() File "/home/lguibert/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1161, in **call** return self.main(*args, **kwargs) File "/home/lguibert/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1082, in main rv = self.invoke(ctx) File "/home/lguibert/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 1443, in invoke return ctx.invoke(self.callback, **ctx.params) File "/home/lguibert/conda/envs/umi/lib/python3.9/site-packages/click/core.py", line 788, in invoke return __callback(*args, **kwargs) File "/home/lguibert/Desktop/UMI_test/universal_manipulation_interface/run_slam_pipeline.py", line 68, in main assert map_path.is_file() AssertionError
> 
> [slam_stderr.txt](https://github.com/user-attachments/files/18800654/slam_stderr.txt) [slam_stdout.txt](https://github.com/user-attachments/files/18800655/slam_stdout.txt)

We did not encounter any issues with the sample data, and we apologize for any inconvenience

Ghr0228, 2025-04-14:
> We have solved this problem, if you have relevant questions welcome to communicate!

Hi,bro,I meet same question,How to communicate with you? I want to know the source of the problem and how do you deal with it?

George-Gwj, 2025-04-24:
可能是一开始录制的时候相机抖动，导致前面十几帧失效，可以试试看一开始录制的时候先静止几秒钟。

s100179, 2025-07-09:
Hi, we also encountered the same problem when running run_slam_pipeline.py.
Could you kindly share how you resolved this issue on your end, or what kind of environment setup you used to collect data?
Thanks!

> We have solved this problem, if you have relevant questions welcome to communicate!

wy-afk, 2025-07-17:
Hi!, I have also encountered problem with mapping. There was no problems with the example set. Can I have a look at the data you originally had and the one with better environment? Thank you !