real-stanford/universal_manipulation_interface #74: 06_generate_dataset_plan:  no valid gripper data

MrXudan, 2025-02-25:
**Is it my problem in data collection or what？Please help me.The following problem was encountered during the execution of step 6:**
############# 06_generate_dataset_plan ###########
Ignored demo_cupC3441326139715_2025.02.12_10.03.21.215050, no camera_trajectory.csv
Ignored demo_cupC3441326139715_2025.02.12_10.03.55.065533, no tag_detection.pkl
Ignored demo_cupC3441326139715_2025.02.12_10.13.56.149350, no tag_detection.pkl
Found following cameras:
camera_serial
C3441326139715    53
Name: count, dtype: int64
Assigned camera_idx: right=0; left=1; non_gripper=2,3...
             camera_serial  ...                                        example_vid
camera_idx                  ...                                                   
0           C3441326139715  ...  demo_cupC3441326139715_2025.02.12_10.02.33.901117

[1 rows x 3 columns]
Traceback (most recent call last):
  File "/home/dhu/universal_manipulation_interface/scripts_slam_pipeline/06_generate_dataset_plan.py", line 787, in <module>
    main()
  File "/home/dhu/anaconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
  File "/home/dhu/anaconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1082, in main
    rv = self.invoke(ctx)
  File "/home/dhu/anaconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/dhu/anaconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 788, in invoke
    return __callback(*args, **kwargs)
  File "/home/dhu/universal_manipulation_interface/scripts_slam_pipeline/06_generate_dataset_plan.py", line 678, in main
    gripper_interp = get_interp1d(gripper_timestamps, gripper_widths)
  File "/home/dhu/universal_manipulation_interface/umi/common/interpolation_util.py", line 10, in get_interp1d
    fill_value=(x[0], x[-1]))
IndexError: list index out of range
Traceback (most recent call last):
  File "/home/dhu/universal_manipulation_interface/run_slam_pipeline.py", line 125, in <module>
    main()
  File "/home/dhu/anaconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1161, in __call__
    return self.main(*args, **kwargs)
  File "/home/dhu/anaconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1082, in main
    rv = self.invoke(ctx)
  File "/home/dhu/anaconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 1443, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/dhu/anaconda3/envs/umi/lib/python3.9/site-packages/click/core.py", line 788, in invoke
    return __callback(*args, **kwargs)
  File "/home/dhu/universal_manipulation_interface/run_slam_pipeline.py", line 121, in main
    assert result.returncode == 0
AssertionError

zhiweicai07, 2025-02-26:
Have you solved this problem? I'm also encountering the same problem.

MrXudan, 2025-02-27:
> Have you solved this problem? I'm also encountering the same problem.

Sadly no :(

zhiweicai07, 2025-02-27:
> > Have you solved this problem? I'm also encountering the same problem.
> 
> Sadly no :(
Can I get your contact information? We can communicate with each other. I'm currently replicating this work. 
Excuse me, how do you connect your UR5 robotic arm to the WSG50-110 gripper?

MrXudan, 2025-02-28:
> > > Have you solved this problem? I'm also encountering the same problem.
> > 
> > 
> > Sadly no :(
> > Can I get your contact information? We can communicate with each other. I'm currently replicating this work.
> > Excuse me, how do you connect your UR5 robotic arm to the WSG50-110 gripper?

I'm also reproducing this project, but only in terms of dataset creation, so I may not be too helpful in terms of the robotic arm and gripper jaws

zhiweicai07, 2025-02-28:
> > > > Have you solved this problem? I'm also encountering the same problem.您解决了这个问题吗？我也遇到同样的问题。
> > > 
> > > 
> > > Sadly no :(  可悲的是不:(
> > > Can I get your contact information? We can communicate with each other. I'm currently replicating this work.我可以获取您的联系信息吗？我们可以互相交流。我目前正在复制这项工作。
> > > Excuse me, how do you connect your UR5 robotic arm to the WSG50-110 gripper?对不起，您如何将UR5机器人手臂连接到WSG50-110 Gripper？
> 
> I'm also reproducing this project, but only in terms of dataset creation, so I may not be too helpful in terms of the robotic arm and gripper jaws我也在复制该项目，但仅在数据集创建方面，所以我可能对机器人的臂和抓爪没有太大帮助

Did you record it with a GOPRO9 camera? I used a GOPRO12. I'm wondering if there is a problem with the camera.

MrXudan, 2025-03-02:
> Did you record it with a GOPRO9 camera? I used a GOPRO12. I'm wondering if there is a problem with the camera.

Yes，recorded it with a GOPRO9 camera but still failed  :(

MarioCavero, 2025-04-12:
Hello @MrXudan @zhiweicai07 . This is the part of the code giving you the error. It seems to me that it's related to the detections you obtain. Also, seeing `Ignored demo_cupC3441326139715_2025.02.12_10.03.21.215050, no camera_trajectory.csv` is more normal but `Ignored demo_cupC3441326139715_2025.02.12_10.03.55.065533, no tag_detection.pkl` ...

```
  gripper_timestamps = list()
  gripper_widths = list()
  for td in tag_detection_results:
      width = get_gripper_width(td['tag_dict'], 
          left_id=left_id, right_id=right_id, 
          nominal_z=nominal_z)
      if width is not None:
          gripper_timestamps.append(td['time'])
          gripper_widths.append(gripper_cal_interp(width))
  gripper_interp = get_interp1d(gripper_timestamps, gripper_widths)
  
  gripper_det_ratio = (len(gripper_widths) / len(tag_detection_results))
  if gripper_det_ratio < 0.9:
      print(f"Warining: {video_dir.name} only {gripper_det_ratio} of gripper tags detected.")
```

 I can also see in the beginning of the script that some files are lacking `tag_detection.pkl`. It's fine to not have camera_trajectory as orbslam3 may have failed, but tag_detection is a different process so `04_detect_aruco` should work. I am wondering if maybe the detect_aruco is not working well with terms of gripper widths. Try plotting the .pkl within the video frames.