real-stanford/universal_manipulation_interface #80: 运行eval_real读取不到视频，出来界面是黑色的，record_robot_world_hand_eye有摄像头画面

Halo-cqu, 2025-03-27:


MarioCavero, 2025-04-12:
Did you connect the micro-hdmi properly to the camera? You should either see the image, or a "black image" with "EL GATO" sign (the connection between capture card and pc is done but something is missing from camera to capture card. @Halo-cqu

huangyan28, 2025-04-12:
I met the same issue as you. But I found it could work randomly, sometimes it could just work. If u have any advice, please let me konw

Halo-cqu, 2025-04-15:
> Did you connect the micro-hdmi properly to the camera? You should either see the image, or a "black image" with "EL GATO" sign (the connection between capture card and pc is done but something is missing from camera to capture card. [@Halo-cqu](https://github.com/Halo-cqu)

you can change umi/real_world/uvc_camera.py

from:

examples['camera_capture_timestamp'] = 0.0
examples['camera_receive_timestamp'] = 0.0
examples['timestamp'] = 0.0
examples['step_idx'] = 0
 
vis_ring_buffer = SharedMemoryRingBuffer.create_from_examples(
    shm_manager=shm_manager,
    examples=examples if vis_transform is None
        else vis_transform(dict(examples)),
    get_max_k=1,
    get_time_budget=0.2,
    put_desired_frequency=capture_fps
)

to:

examples['camera_capture_timestamp'] = 0.0
examples['camera_receive_timestamp'] = 0.0
examples['timestamp'] = 0.0
examples['step_idx'] = 0
 
vis_examples = copy.deepcopy(examples)
 
if vis_transform is not None:
    vis_shape = (720, 960, 3)
else:
    vis_shape = shape+(3,)
vis_examples['color'] = np.empty(shape=vis_shape, dtype=np.uint8)
vis_ring_buffer = SharedMemoryRingBuffer.create_from_examples(
    shm_manager=shm_manager,
    examples=vis_examples,
    get_max_k=1,
    get_time_budget=0.2,
    put_desired_frequency=capture_fps
)

Halo-cqu, 2025-04-15:
> I met the same issue as you. But I found it could work randomly, sometimes it could just work. If u have any advice, please let me konw

you can change umi/real_world/uvc_camera.py

from:

examples['camera_capture_timestamp'] = 0.0
examples['camera_receive_timestamp'] = 0.0
examples['timestamp'] = 0.0
examples['step_idx'] = 0

vis_ring_buffer = SharedMemoryRingBuffer.create_from_examples(
shm_manager=shm_manager,
examples=examples if vis_transform is None
else vis_transform(dict(examples)),
get_max_k=1,
get_time_budget=0.2,
put_desired_frequency=capture_fps
)

to:

examples['camera_capture_timestamp'] = 0.0
examples['camera_receive_timestamp'] = 0.0
examples['timestamp'] = 0.0
examples['step_idx'] = 0

vis_examples = copy.deepcopy(examples)

if vis_transform is not None:
vis_shape = (720, 960, 3)
else:
vis_shape = shape+(3,)
vis_examples['color'] = np.empty(shape=vis_shape, dtype=np.uint8)
vis_ring_buffer = SharedMemoryRingBuffer.create_from_examples(
shm_manager=shm_manager,
examples=vis_examples,
get_max_k=1,
get_time_budget=0.2,
put_desired_frequency=capture_fps
)